LinuxCNC/linuxcnc #3463: Why build (almost) empty translations?

BsAtHome, 2025-06-06:
Many translations are built when you do, for example, make docs. But most are effectively not translated at all. A list, extracted from po4a messages and summed shows the following:
```
lang #translated #total relative (weblate)
ar   207 41872  0.49% (1%)
bg   138 41876  0.33% (1%)
ckb  138 41876  0.33% (1%)
cs   139 41876  0.33% (2%)
da   138 41876  0.33% (1%)
de 33712 38283 88.06% (98%)
es   508 41874  1.21% (11%)
fr  1118 41845  2.67% (10%)
hu   219 41874  0.52% (3%)
it   168 41876  0.40% (11%)
ka  2667 41876  6.37% (1%)
nb  4900 41845 11.71% (9%)
pl   138 41876  0.33% (3%)
pt   200 41845  0.48% (1%)
pt_BR 232 41874 0.55% (4%)
ro   213 41866  0.51% (2%)
ru 11810 41532 28.44% (35%)
sai  138 41876  0.33% (1%)
sk   194 41876  0.46% (11%)
sv   167 41876  0.40% (1%)
tr   328 41845  0.78% (1%)
uk   255 41876  0.61% (1%)
vi   149 41876  0.36% (5%)
zh_CN 337 41845 0.81% (10% or 28%)
```
The lowest number, 138, originates from the fact that one common string is reused many times (the 'en' language tag into the localized version).
The differences between my tally and the Weblate version is due to the fact that I did not reintegrate Weblate's translations. Therefore, the separate column. But, notably, Finnish (fi) is missing in LCNC (but is at 1%, thus effectively untranslated). It should also be noted that there are three Chinese translations in Weblate and only one in LCNC and I have no idea which is actually mapped

In summary, 10 of the the 24 translations are ballast and completely unusable. There is no (usable) translation in Weblate and reintegration into LCNC has no benefit. There has not been activity on these translations and the only really active and up-to-date translation is German.

My suggestion is to trim the list and remove ar, bg, ckb, da, ka, pt, sai, sv and uk (all under 2%).

Please comment.

andypugh, 2025-06-07:
Are you suggesting that we drop them from weblate, or just filter them out of the documentation build? 

Assuming the latter, how would we re-introduce them if someone did submit a significant quantity of translation? (ie, how would we notice, and who would remember that this needed to be done?)

snowgoer540, 2025-06-16:
> Please comment.

I've suggested this before, and I will suggest it again.  I think these translations should be their own separate project.  I am not sure the logistics of that, but this current system, while perhaps good intentioned, is not working.

To expand further, I think what happened is a few years ago a few people got together and decided they were going to champion the docs.  As it goes, it is quite the unglamourous job, and people were happy for them to comb through the docs and fix whatever they saw.  Commit rights were extended to new people (who now seem to have disappeared).  Eventually the bulk majority of the commits for a long stretch of time were solely with regard to straightening out the documentation and adding translations and its supporting build structure.  It got to the point where commits would regularly break the buildbot(s), or sometimes LinuxCNC itself (version changes without warning, etc. which prevented LinuxCNC from starting), but it was tolerated for some reason because working on the docs is unglamorous (I guess?)... In the end it seems to have caused more grief for the people who use the project than those who couldn't do without the translations.  It's made the size of the project bloated, it's completely ruined searching in the base directory, and as you said, it doesn't work.

I can tell you for a fact that people have left the project over not only the commits but the attitude of those making the commits.  It seemed like the documentation translations became more important than the project itself.  

I say all of that to say it doesn't bother me in the least if you remove any of this from the project.  All that should stay are the source files, in English.  It's the official language of air traffic control across the world, why can't it be the official language of LinuxCNC as well.  Having written my share of documentation for this project, I can tell you that most don't read it anyways.  It's easier to ask questions on the forum, so that's what people do.

Lastly, PDF is a broken mess.  It either needs to be completely revamped, or gutted.

BsAtHome, 2025-06-16:
Thanks for you comment :-)

I do recognize the appeal of splitting the documentation into separate projects, but it is a bit of a pain doing so.
The manpages, a major part of the documentation, cannot be split into a separate project. These are a) needed for building the deb package and b) quite a few are generated from the code-files (components embed docu extracted using halcompile).
So, splitting the docs would only involve a part making things harder.

The current build system only makes documentation debs for `en`  and `de`. The rest is discarded after the build to get out of the way. It looks like the only parts that make it into the deb doc packages are some PDFs. That would leave us in the situation where RIP builds are the *only* (semi) effective way to read localized documentation, if you can find the files.

I'm not even sure how the manpage translations are handled when we are looking at packaging. I can see that there are two gettext .mo bindings (gmoccapy and linuxcnc) for the code, but am unsure what *exact* code actually binds to what. FWIW, I don't even know what is the best practice how to distribute manpage translation as Debian packages. Maybe @smoe can give me a clue what Debian expects when you want to distribute translated manpages? This is distinctly different from the documentation packages, which can be either html or pdf, but only pdf is currently used.
Not sure how or if html doc is distributed. I have a slight feeling that it requires some <strike>divine</strike>manual intervention to have the html on linuxcnc.org to be up to date (maybe @andypugh can tell us how that is done).

> I say all of that to say it doesn't bother me in the least if you remove any of this from the project. All that should stay are the source files, in English. It's the official language of air traffic control across the world, why can't it be the official language of LinuxCNC as well.

Not everybody is good at English. Most developers, yes, but operators? A real problem are the (mill-, lath-, cutter-,...) technical terms, which can be problematic if presented in English to a non-English operator. Translations are a necessity, it is as simple as that. Saying to drop everything except English does not really help to achieve world wide <strike>domination</strike>adoption. ;-)

Anyway, also feel free to add comments to #3468 too if you have an idea.

smoe, 2025-06-17:
Thinking more about it - maybe there is no need for our translations to surface in _any_ of our public repositories. Well, weblate may have a say on this, too, but there may be a way to have this addressed by a hidden mirror repository of the English original.

We could then continue redistributing the English documentation as a package (no idea why exactly, but there is a demand for it https://qa.debian.org/popcon.php?package=linuxcnc ) and would distribute the others only via our website. I would still suggest to have the documentation as project of its own, though. I am not exactly sure how such a separation would change the nature of the documentation. It seems likely that differences between versions would not only describe the latest behaviour but also describe how certain interactions have been done before. That complicates things a bit?

The documentation of LinuxCNC IMHO is very important for the project since it is very likely to be the very first encounter of anyone deciding to invest further into that technology. I have bought the https://www.amazon.de/Hobbyists-Guide-CNC-Dennis-Grant book and this only talks about Mach4 on Windows, mentions Linux only as something of too much as a hurdle for someone who primarily wants to chip some metal. How dare he. We have YouTube videos, the LiveCD/USB stick and our documentation to set this record straight - hey, and why not also distribute our documentation as a book, well, hm, this needs some extra efforts, but I am still confident that it would help our project to get some sort of a book out that starts out with what our documentation already offers.

andypugh, 2025-06-18:
> hey, and why not also distribute our documentation as a book

https://www.amazon.co.uk/LinuxCNC-Getting-Started-Guide-Team/dp/9888406302

(I have no idea who published this, or where the money goes, it's not us) 
I would set the record straight, but when I try to leave a review I get "We apologize but Amazon is not accepting reviews on this product from this account."