LinuxCNC/linuxcnc #3464: Axis Gui: MDI command stacking broken in currrent master

Sigma1912, 2025-06-07:
Looks like something has broken MDI command stacking in the AxisGui in current master.

[image]
 

New commands can only be added after the previous one has been executed.

This works in 2.9.4

andypugh, 2025-06-07:
https://github.com/LinuxCNC/linuxcnc/issues/2956 was a request to prevent queuing, but seems to still be open.

Sigma1912, 2025-06-07:
Actually it seems that there is already an open issue about this for 2.9:
https://github.com/LinuxCNC/linuxcnc/issues/2460

which was apparently then fixed in 2.9 with this commit:
https://github.com/LinuxCNC/linuxcnc/commit/7852337dfdc6cda49db31a3b13ca333d4e6d874f

But it seems that it has not been merged into master:
https://github.com/LinuxCNC/linuxcnc/blob/master/src/emc/usr_intf/axis/scripts/axis.py#L3262

Patching my build of current master with the changes in the commit above fixes this issue.

Sigma1912, 2025-06-07:
I think that a change to prevent mdi command queuing would require for this to be configurable in the ini. There are going to be a lot of unhappy users if it were just simply changed from one to the other. 
Looking at this issue makes me think that this may even be possible to implement on the GUI level.

andypugh, 2025-06-07:
I can probably just cherry-pick 7852337dfdc6cda49db31a3b13ca333d4e6d874f

Let me try

andypugh, 2025-06-07:
> I think that a change to prevent mdi command queuing would require for this to be configurable in the ini. 

I have a feeling that it is. 
See the comment about MDI_QUEUED_COMMANDS here: 
https://github.com/LinuxCNC/linuxcnc/issues/2460#issuecomment-1537248558

Need to experiment.

andypugh, 2025-06-07:
MDI_QUEUED_COMMANDS is undocumented and has an odd effect. 
Set to 1 it seems to make no difference, I could still queue 3 commands. 
Set to 0 it completely disables MDI. 
So it looks like it sort-of works, but isn't working right. (probably why it is undocumented)

Sigma1912, 2025-06-08:
- In 'taskmain.cc' the value of MDI_QUEUED_COMMANDS seems not to be actually used for anything.

- In 'axis it is used 
```
    return s.interp_state == linuxcnc.INTERP_IDLE or (s.task_mode == linuxcnc.MODE_MDI and s.queued_mdi_commands < vars.max_queued_mdi_commands.get())
```
but the problem here is that the attribute 'queued_mdi_commands' in the python status channel seems to always be zero:

[image]

Sigma1912, 2025-06-08:
strangely though there seems to be a test for 'queued_mdi_commands':
```
./tests/mdi-queue-length/test-ui.py:51:        if s.queued_mdi_commands == queue_len:
./tests/mdi-queue-length/test-ui.py:54:    print("queued_mdi_commands at %d after %.3f seconds" % (s.queued_mdi_commands, timeout))
./tests/mdi-queue-length/test-ui.py:85:assert(s.queued_mdi_commands == 0)
./tests/mdi-queue-length/test-ui.py:93:assert(s.queued_mdi_commands == 0)
./tests/mdi-queue-length/test-ui.py:99:assert(s.queued_mdi_commands == 1)
./tests/mdi-queue-length/test-ui.py:106:assert(s.queued_mdi_commands == 2)
./tests/mdi-queue-length/test-ui.py:111:assert(s.queued_mdi_commands == 3)

```

andypugh, 2025-06-09:
Am I right in inferring that MDI queuing in this context is limited to the queing of queue-busters? So there is an MDI-queue separate from the motion queue?

Sigma1912, 2025-06-09:
No, it is not limited to the queing of queue-busters. The command c.mdi('m66 p0 l3 q30') in the test is to keep motion busy for 30 seconds and then other mdi commands are issued which are added to the mdi_queue until motion has finished waiting for 30 seconds and empties the other commands from the mdi_queue:
https://github.com/LinuxCNC/linuxcnc/blob/master/tests/mdi-queue-length/test-ui.py#L87 

Without the 'm66 p0 l3 q30' the mdi commands would go straight into the motion queue which is why I thought  'queued_mdi_commands' was not working. 
So in short if an mdi command cannot be added to the motion queue it gets put in the separate mdi_queue.