LinuxCNC/linuxcnc #3468: RFC: Rework documentation and translation

BsAtHome, 2025-06-09:
Many small things were seen and discovered while trying to understand how the documentation tree is built is how translations are created. The current build system has a very long history and has grown may heads like a hydra. And while cutting one head, two grew back.

The documentation subsystem is a very important part of LinuxCNC and there is, very fortunately, a lot of documentation. But it requires a caring hand to make it simpler and more flexible to handle and to become easier to build and maintain.

* Currently, manual html files are created from troff formatted files instead if the adoc sources. The html files must be generated from the asciidoc sources. The problem are RT components, where halcompile cannot generate adoc formatted documentation. Halcompile needs to be adapted and the component files need to be adjusted to use adoc.
**question**: we should drop halcompile troff generation. It would be a better to do so because generating both may pose severe problems (like tables). You can still generate troff from adoc, but using halcompile to "view" the documentation is not be necessary.
 
* There seem to be two ways how translations are put together in the makefile. One is (part of) `es` and then the rest. It seems to be a remnant of other times, but it complicates things when no consistent way of translation building and organization is adhered to.
 
* All translations should be handled in a sub-directory structure instead of all over the place. Currently, multiple language name directories are created under docs, docs/src and more. Translations should be created under one directory. Such structure allows for one makefile rule to descend via a sub-make into each directory and do the same operations to build a set of documents in the sub-tree:
```
docs - xlate + en + man  + man1
             |    |      + man3
             |    |      + man9
             |    |
             |    + html + images
             |    |
             |    + src  + code
             |           + common
             |           + config
             |           + driver
             |           + ...
             |
             + de + man  + man1
             |    |      + man3
             |    |      + man9
             |    |
             |    + html + images
             |    |
             |    + src  + code
             |           + common
             |           + config
             |           + driver
             |           + ...
             + es + man  + man1
            ... ... ...
```
* Note that `en` is a language in the sub-tree as it is used to build the docs in that tree. Each language, even the source language, should be treated equally to make one rule to handle all languages. It might also make inserting generated docs easier if the source language tree is first built and then the translations are taken from that specific generated tree into the localized versions.

* Using adoc as intermediate format for component man-pages might complicate translation (there seem to be *some* translations today, but inconsitently).
**question**: How can we assure easily and consistently translated component documentation when the sources for translation are an intermediate format? If the comp-file sources are to be used as master, then how would the data be extracted and merged?
 
* Localized images for the documents are a separate issue. Po4a can be easily adapted/configured to use the above directory structure. However, images need to be created for each language and redistributed into the build-tree. There are some localized images currently in the sources. Using these needs to be standardized.
 
* Is it really necessary to build PDF documentation? Can't we suffice with HTML? Or is it a natural consequence of adoc that you want and epub format published? (and the man-pages PDF is not doing a proper job of highlighting because it is built from troff data instead of adoc sources; is also misses TOC, index and cross-reference)
 
* Characters are missing when building PDFs (notably ar and zh_CN). Apparently, some fonts/glyphs were not available when they were created. That indicates missing dependencies in my debian12 VM. Does anybody know which dependencies?

* There does not seem to be one way to format documents. Not that it is a big problem, but for parts you'd like consistency. For example, you want all man-pages to be able to link through the "name(X)" references between each other. But also the look-and-feel should be kept similar. This formatting also applies to the adoc sources.
**question**: are there currently formal format guidelines?

* Feel free to add to the list.

--
A different story, and a really hard one, is to reorganize a lot of documentation *content*. A lot of information is almost impossible to find unless you know where to find it. A lot of files cover many, many topics. Some may be duplicated and others are in illogical order. The documentation must clearly distinguish between introduction(s), explanation(s), procedure(s) and reference(s).

All of this must be organized with several types of users in mind so that each type can find the relevant information in a blink of the eye.

Some observations (from a pedagogical perspective):
- Introduction(s) that include bewildering details. Concepts jump from simple explanations into details about programming. For example, a split between gcode programming and LCNC interpretation is in order.
- Many paragraphs with headers and one line of content. Many do not convey (enough) information or are redundant or the information is obvious.
- A whole part of plasma cutting is inserted (2.7) and it goes from a "primer" and further into details with a specific hardware card. This is specifically an adaption for plasma cutting that would warrant a separate document. Especially because chapter 3 then goes into configuration wizards (and then seems to forget to mention Mesa Configuration Tool II). Therefore, the order of information presentation is not good (intermixes several levels of principles, implementations and applications).
- (and much more)

There needs to be a way to make things easy for beginners and not to belittle experienced users. And for this a strategy must be made and a clear path should be visible throughout the documentation.

Don't despair. The good thing,... most documentation is already there. It just needs to be put in the right order.

Sigma1912, 2025-06-09:
What you propose sounds great but where would we get the personnel from to rework everything? Surely that would be a monumental task even for a well organized project. The way things work here I'm afraid this would take years to accomplish and will more likely peter out somewhere along the way. We barely manage to keep up with what we already have. 
Take this example of an attempted small fix in a document: https://github.com/LinuxCNC/linuxcnc/pull/3448 After 2 weeks it's still a work in progress, mostly because people start a discussion and then disappear. The issues section is full of stuff like that. Even in the Pull Request section there are stalled discussions because people just disappear.

BsAtHome, 2025-06-11:
> Would you just move paragraphs around or are you expecting others to rewrite their part of the documentation?

Something in that direction. Move stuff to become more accessible and have people contribute. It is a lot easier for anyone to contribute when it is accessible. But you probably need to invite people to write small parts. And, being _small parts_ would make the likelihood of success increase.

Sigma1912, 2025-06-11:
That sounds like a fairly realistic take on things.

smoe, 2025-06-17:
I see several layers:
 * overview on how LinuxCNC works and how it works with other tools to get work done
 * get it all installed, configured, running and monitored
 * extend it
 * change it

The resources available (or parts of it) it would assign to one of these layers. These layers then get different priorities and the more abstract bits likely are more narrative than the lower-level techie bits.

BsAtHome, 2025-06-17:
Access to weblate maintenance apparently requires some extra setting. @smoe, I can see you are maintainer. Can you tell me how the components are actually configured, including all hooks? Or give me access to that information?

petterreinholdtsen, 2025-07-06:
Dear BsAtHome and others.

I like your thoughts on the documentation and translations.  I was the
one adding the current PO based translation framework into LinuxCNC, and
you are absolutely right that the build system is not currently a well
designed an functioning machine.  I got it limping along, and ran out of
time to improve it further.  If you got time to improve it, please go
ahead.  Note, there are talks on converting linuxcnc to using cmake, and
even a draft for this in some branch somewhere.  Sadly the cmake build
rules do not support building the translated documenation, so it need
some more work before it can replace the current automake based build
system.

I agree that it would be better if components included adoc fragments
instead of groff fragments, making all man pages extracted by halcompile
adoc files.  This will most likely take a few hours to do.  Note, I
believe halcompile need to support both formats for a while (two major
releases?) to keep working with third party components.

You mention that the spanish (es) translation is handled differently
than the other translations in the makefile.  If so, this is by mistake
and not by design, as all translations should be handled the same way.

I agree that a different directory structure with one subdirectory per
translation might be easier.  The reason I did not do this on my
conversion was I did not want to touch the man page building machinery
due to lack of time and a different focus.

You ask how we can handle component documentation via a intermediate
format.  As far as I can tell po4a handle this just fine, even sharing
strings between adoc and roff files if there are no formatting
statements in the strings.  But this will be less of a problem if we
convert halcompile into generating adoc files instead.

You ask if building PDFs is really necessary.  I guess not, but I find
it very convenient to have the documentation both available as one large
searchable PDF and HTML pages for easier web reference.  It also ensure
that we avoid embedding HTML in our documentation, ensure the
documentation source is independent of the presentation format.

I do not know what is needed to get arabic and chinese PDFs building
properly.

I believe there is a draft recommendation on how to format documents.
Steffen was looking into this a few years ago.

-- 
Happy hacking
Petter Reinholdtsen

petterreinholdtsen, 2025-07-06:
I forgot to mention, that if you want to discuss these and other topics,
I would be happy to be approached on IRC (#linuxcnc or #linuxcnc-devel
on chat.libera.chat).  We can even coordinate a discussion via Signal or
a video chat about these topics. :)

-- 
Happy hacking
Petter Reinholdtsen

smoe, 2025-07-06:
The challenge is to maintain the momentum. :-)

We have now some first experiences with what seems like an automated translation (Tamil) which triggered a series of issues with po4a. We should look at this a bit more - frankly, our human time should preferably go into what is difficult to translate and/or figures and preferably into more and better source code.

BsAtHome, 2025-07-06:
> frankly, our human time should preferably go into what is difficult to translate and/or figures and preferably into more and better source code.

That is terribly short-sighted!

There is no point in "better source code" when the documentation is bad, missing, misleading or outdated.

smoe, 2025-07-06:
> > frankly, our human time should preferably go into what is difficult to translate and/or figures and preferably into more and better source code.
> 
> That is terribly short-sighted!
> 
> There is no point in "better source code" when the documentation is bad, missing, misleading or outdated.

Hm. Those interested please find the complete sentence of mine above. I stand to it. We have a technology now that does a very good job to translate for us. And I strongly suggest we employ that technology where we see fit - and improve that technology (or make some noise for others to improve it) where that technology fails. How exactly it is to be used, and if that is applicable for all language, that is up for debate.