LinuxCNC/linuxcnc #3497: Gmoccapy - duplicate parameters in *.pref and *.ini files.

zz912, 2025-07-04:
This Issue is a continuation of the discussion from here:
https://github.com/LinuxCNC/linuxcnc/pull/3488

Gmoccapy historically reads some parameters from a *.ini file, then saves them to a *.pref file. Then it reads and changes them using the Setup Page.
It is frustrating when you set parameters from a *.ini file, start Gmoccapy, the parameters are read correctly, but subsequent changes to the *.ini file are ignored. I have always struggled with this, but I thought it was just my distraction.

Currently, ISTAT check integration is in progress and there are plans to implement check INI parameters.
http://linuxcnc.org/docs/devel/html/gui/gui-dev-reference.html

It is necessary to decide how to deal with duplicate parameters in the Setup Page (*.pref file) in the future.

zz912, 2025-07-04:
> > some parameters from a *.ini file, then saves them to a *.pref file.
> 
> Could you indicate which parameters these are?

For example:

1)
**DEFAULT_SPINDLE_SPEED:**
here is readed during first run Gmoccapy:  https://github.com/LinuxCNC/linuxcnc/blob/259ad89a34a86bccb51d7fa0e5f2830c734822c4/src/emc/usr_intf/gmoccapy/getiniinfo.py#L281
here is saved to pref file:
https://github.com/LinuxCNC/linuxcnc/blob/259ad89a34a86bccb51d7fa0e5f2830c734822c4/src/emc/usr_intf/gmoccapy/gmoccapy.py#L4330
(I have verified here that this parameter behaves as I wrote above)

2)
in future ISTAT will check:
**MAX_SPINDLE_SPEED**
But in *.pref exists spindle_bar_max:
https://github.com/LinuxCNC/linuxcnc/blob/259ad89a34a86bccb51d7fa0e5f2830c734822c4/src/emc/usr_intf/gmoccapy/gmoccapy.py#L4339

At this moment, I don't finding out how many parameters there are, because the rules for ISTAT are not yet set and I don't know how Norbert will approach these parameters.

This Issue is not about a technical solution to the problem, but about how the "Setup Page" in Gmoccapy should be developed.
[image]

> Is there a particular reason for not writing these values to the ini file instead?

https://github.com/LinuxCNC/linuxcnc/pull/3488#issuecomment-3026649588

>  After all that is what the 'calibration' tool has been doing for years.

What is 'calibration' tool?

Sigma1912, 2025-07-04:
> What is 'calibration' tool?

The tool to set the JOINT ini values:

[image]

[image]

Sigma1912, 2025-07-04:
I would tend to just simply remove the 'Spindle' box from the Settings page. 
In any case, having the pref file overrule an ini file setting seems wrong to me.

Sigma1912, 2025-07-04:
Thank you for your comments

> Why does the location matter so much? It's very clear where it is from the source.

It's not that it's hard to find. The point is that a folder structure should order files in a logical manner.  There are of course different ways of doing this but looking at the 'lib/python' folder it seems that the original idea was to have the files structured by gui or application. If this is to make any sense we must not have things in one gui folder import stuff from another gui folder. The things that are to be shared should be in a separate (eg 'common') folder. 
When I maintain a GTK based gui like Gmoccapy I should _never_ have to deal with something in a folder called 'qtvcp',  'touchy', 'stepconf' or 'plasmac'.

Just imagine some future person trying to remove one of those guis or applications from the source code. If the folder 'qtvcp' is removed the Gtk based Gui called Gmoccapy will no longer work. This is a clear indication that the folder structure has been broken and can no longer serve its purpose.

zz912, 2025-07-04:
> Why does the location matter so much? It's very clear where it is from the source.
> I'm asking these relentless questions so I make better choices/changes in the future.
> To me it seems a bit OCD - which is fine, there are plenty of things I do just because it 'feels' right.

I am an amateur programmer, self-taught. I learned Python, Glade, Git only to help with LCNC maintenance. I am only interested in Gmoccapy, Mesa 7i96 + doughter cards, PncConf. When I saw QT in Gmoccapy, I was scared that I would have to learn another technology, even though I know the others only minimally. I didn't want to hinder development, but if it is possible to continue maintaining/improving Gmoccapy without other new technologies for me, it will help me.

c-morley, 2025-07-04:
> When I maintain a GTK based gui like Gmoccapy I should never have to deal with something in a folder 

> called 'qtvcp', 'touchy', 'stepconf' or 'plasmac'.

I guess this hard line just doesn't apply in my mind. Qtvcp use hal_glib which is gtk/GObject based.
I did this on purpose so as not to double the effort creating a PyQt version of hal_glib.
It also helps that whenever anyone fixes/improves hal_glib all the projects that use it are fixed/improved.
The docs for libraries Gstat (GladeVCP ) and STATUS (Qtvcp) signals and functions are virtually the same.
Much easier for a user/developer to move between the two toolkits.

Sigma1912, 2025-07-05:
> Where should the common flles go ? python/lib/common?
yes

c-morley, 2025-07-05:
Pushed some work on this. Still a bit more to do on iniinfo IStat for path work