LinuxCNC/linuxcnc #3505: gmoccapy won't start on latest version of master

pippin88, 2025-07-06:
gmoccapy won't start on latest 2 versions of master.

Version _1:2.10.0~pre0.4739.gc43a18ff38_ starts without issue

**Problem:**
```
LINUXCNC - 2.10.0-pre0-4751-ge78cc0ac06
Error in sys.excepthook:
Traceback (most recent call last):
  File "/usr/bin/gmoccapy", line 62, in excepthook
    LOG.error(message)
    ^^^
NameError: name 'LOG' is not defined

Original exception was:
Traceback (most recent call last):
  File "/usr/bin/gmoccapy", line 6094, in <module>
    from common import logger
ModuleNotFoundError: No module named 'common'
5952
5962
Stopping realtime threads
Unloading hal components
Note: Using POSIX realtime
```

I encountered this because I re-installed whole system, moving from Debian 10 which was not getting any update for a long time, to now Debian 12. I could not get LinuxCNC to start (I use gmoccapy). Eventually worked out Axis would start. Forcing Synaptic to downgrade to LinuxCNC 2 versions old allows gmoccapy to start and CNC machine responds.

[gmoccapy error.txt](https://github.com/user-attachments/files/21087608/gmoccapy.error.txt)

pippin88, 2025-07-06:
I am not sure if this is some issue with how I updated to Master branch.

I installed latest 2.9.4 ISO.
Updated system via apt.

Then followed https://www.forum.linuxcnc.org/9-installing-linuxcnc/55729-script-to-update-2-9-x-iso-to-use-linuxcnc-version-2-10-master-branch
I used the commands from rodw's script on the first post.
I copied each command and executed separately.

hansu, 2025-07-06:
Maybe one commit in between on master didn't work because Chris worked on the logger.
Maybe the buildbot wasn't that fast and you didn't get the latest one.
I tested the current HEAD: e78cc0ac0679bc7d02fcef89332a4d6164ab6c56 and this works...
I suggest to wait a bit until the buildbot has finished buiding a package for that one.

zz912, 2025-07-07:
I have similar problem.

If I download the current master branch from Github using a zip file, Gmoccapy works.

I did a "git pull" on the cloned folder and gmoccapy no longer started.

```
[Gmoccapy][ERROR]  Found an error!
The following information may be useful in troubleshooting:

Traceback (most recent call last):
  File "/home/zdenek/linuxcnc/linuxcnc-zz912/bin/gmoccapy", line 6136, in <module>
    inifile = sys.argv[2]
              ~~~~~~~~^^^
IndexError: list index out of range
 (gmoccapy:62)
```

I tried: 
make clean
./autogen.sh
./configure --with-realtime=uspace
make -B

but nothing works

c-morley, 2025-07-07:
> gmoccapy won't start on latest 2 versions of master.
>
> Version _1:2.10.0~pre0.4739.gc43a18ff38_ starts without issue
>
> **Problem:**
>
> ```
> LINUXCNC - 2.10.0-pre0-4751-ge78cc0ac06
> Error in [...]

If using an installed version of linuxcnc, it failed because I missed updating a file.
I have since pushed that file several hours ago, so should work now.
Please advice if there are continued problems.

An RIP version would have been fine.

pippin88, 2025-07-09:
Updated to pre0.4764.g9bfdb69cba via apt

Working

(Had to delete my old pref file)

zz912, 2025-07-12:
I think this issue can be closed.

closed 2025-07-12 by andypugh