LinuxCNC/linuxcnc #3513: Gmoccapy offset page - buttons

hansu, 2025-07-09:
Just pull this out of #3494.

[@Sigma1912]
> I have also prepared adding a buttonbox to the offsetpage:
> 
> ![offsetpage](https://private-user-images.githubusercontent.com/46067220/463238443-96a94c66-21c9-4430-97f3-83fc38a6cebb.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTIwNDI3MjUsIm5iZiI6MTc1MjA0MjQyNSwicGF0aCI6Ii80NjA2NzIyMC80NjMyMzg0NDMtOTZhOTRjNjYtMjFjOS00NDMwLTk3ZjMtODNmYzM4YTZjZWJiLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTA3MDklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwNzA5VDA2MjcwNVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWEzYmFmYjFmZDk4MDg1Zjk4MmUzY2NmOTcyMmU0ZDUyNTA5NTVhYzQ4ODRmMGM4NWU1MzY0MjUxZGUxOGM3MDgmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.MVUsGxXj2BGscxZw61mJ5xAG24bNWp3TUbbUuZvZ5Bo)
> 
> Maybe we could move some of the 'edit offsets', 'Zero G92' , 'Set selected offset' buttons to the new button box. What do you think?

hansu, 2025-07-09:
[Sigma1912]
> Proposed tooltable 'change to highlighted': ![Recording](https://private-user-images.githubusercontent.com/46067220/463711913-5ff163e1-541a-4625-b022-92ae35ab3443.gif?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTIwNDI3MjUsIm5iZiI6MTc1MjA0MjQyNSwicGF0aCI6Ii80NjA2NzIyMC80NjM3MTE5MTMtNWZmMTYzZTEtNTQxYS00NjI1LWIwMjItOTJhZTM1YWIzNDQzLmdpZj9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTA3MDklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwNzA5VDA2MjcwNVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWIzNTJhZTA3YTAzMGYzZjVjZTYwMDE5NThiODZmM2M5NDRjODI2NTk2ZWE5MjgxMTYyOTkwYjVkMWUxMzVhNTImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.whtXRa2Y8PZp8PGsvugSuFbAvKDJ6LAZ4LTN_EY9PdU) [ ![Recording](https://private-user-images.githubusercontent.com/46067220/463711913-5ff163e1-541a-4625-b022-92ae35ab3443.gif?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTIwNDI3MjUsIm5iZiI6MTc1MjA0MjQyNSwicGF0aCI6Ii80NjA2NzIyMC80NjM3MTE5MTMtNWZmMTYzZTEtNTQxYS00NjI1LWIwMjItOTJhZTM1YWIzNDQzLmdpZj9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTA3MDklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwNzA5VDA2MjcwNVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWIzNTJhZTA3YTAzMGYzZjVjZTYwMDE5NThiODZmM2M5NDRjODI2NTk2ZWE5MjgxMTYyOTkwYjVkMWUxMzVhNTImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.whtXRa2Y8PZp8PGsvugSuFbAvKDJ6LAZ4LTN_EY9PdU) ](https://private-user-images.githubusercontent.com/46067220/463711913-5ff163e1-541a-4625-b022-92ae35ab3443.gif?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTIwNDI3MjUsIm5iZiI6MTc1MjA0MjQyNSwicGF0aCI6Ii80NjA2NzIyMC80NjM3MTE5MTMtNWZmMTYzZTEtNTQxYS00NjI1LWIwMjItOTJhZTM1YWIzNDQzLmdpZj9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTA3MDklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwNzA5VDA2MjcwNVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWIzNTJhZTA3YTAzMGYzZjVjZTYwMDE5NThiODZmM2M5NDRjODI2NTk2ZWE5MjgxMTYyOTkwYjVkMWUxMzVhNTImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.whtXRa2Y8PZp8PGsvugSuFbAvKDJ6LAZ4LTN_EY9PdU) [ ](https://private-user-images.githubusercontent.com/46067220/463711913-5ff163e1-541a-4625-b022-92ae35ab3443.gif?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTIwNDI3MjUsIm5iZiI6MTc1MjA0MjQyNSwicGF0aCI6Ii80NjA2NzIyMC80NjM3MTE5MTMtNWZmMTYzZTEtNTQxYS00NjI1LWIwMjItOTJhZTM1YWIzNDQzLmdpZj9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTA3MDklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwNzA5VDA2MjcwNVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWIzNTJhZTA3YTAzMGYzZjVjZTYwMDE5NThiODZmM2M5NDRjODI2NTk2ZWE5MjgxMTYyOTkwYjVkMWUxMzVhNTImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.whtXRa2Y8PZp8PGsvugSuFbAvKDJ6LAZ4LTN_EY9PdU)
> 
> Todos:
> 
>     * remove automatically checking checkbox of currently loaded tool
> 
>     * maybe show currently loaded too in the new buttonbox as tool info frame is hidden when using the onboard keyboard.
> 
> 
> Comments welcome.

Sigma1912, 2025-07-13:
I've moved away from trying to add more buttons to the bottom row. I think it's preferable to have the buttons that directly relate to editing values in the table inside the table frame, just like in the tool editor https://github.com/LinuxCNC/linuxcnc/pull/3509.

Accordingly I would suggest to move the 'calculator' and 'zero g92' to the table frame and hide those until the 'edit offset' button is activated

Sigma1912, 2025-07-14:
> Fine for me, but some users have hardware buttons beside the screen for the lower and right button box...

I know, that was the original layout but we've always had buttons in the preview window that are not aligned with that design so we have never really been consistent in that respect. 
But you are right it may be worth to rethink the whole idea a bit. 

For the use of Gmoccapy w/o touch screen an aligned hardware button doesn't really make life much easier if the assigned function requires mouse movement. I might as well call that function with the mouse in the first place. 
So, say I have aligned hardware buttons and a keyboard. Would I really want to use

1. the hardware button to activate the 'edit offsets' function
2. the mouse to select a cell in the offset table
3. the keyboard to enter a value
4. the hardware button to deactivate the 'edit offsets' function

So in that sense we could move  the 'edit offsets', 'Set selected offset' and 'use calculator' buttons to the table frame without making gui operation in any way less convenient for users with aligned hardware buttons. For everybody else it really doesn't matter anyway as long as the grouping of the buttons still makes some logical sense.

> But "zero g92" should be also clickable when not in edit mode.

Yes, I agree and it should remain in the bottom row of buttons. Although we may want to make it clearer that this also zeros G52 as these two offsets share the same values.

hansu, 2026-01-07:
The problem is only that the offsets just fit into the available height when running on 1024x768.
With the buttons you would need to scroll which I would like to avoid.

- We could hide the lower part with Tool info, Velocity and Spindle speed info and expand the offsetpage to the bottom.
- We could hide the Jog area and add the buttons there

But both is not really ideal I think.

[image]

closed 2026-01-21 by hansu