LinuxCNC/linuxcnc #3520: CI on 2.9 fails - Debian Buster archived

BsAtHome, 2025-07-19:
It seems that Debian Buster (Debian 10) has been archived and is no longer available from the default repos. 

In 'Install pre-dependencies' stage:
```
Err:4 http://deb.debian.org/debian buster Release
  404  Not Found [IP: 151.101.202.132 80]
Err:5 http://deb.debian.org/debian-security buster/updates Release
  404  Not Found [IP: 151.101.202.132 80]
Err:6 http://deb.debian.org/debian buster-updates Release
  404  Not Found [IP: 151.101.202.132 80]
Reading package lists...
E: The repository 'http://deb.debian.org/debian buster Release' does not have a Release file.
E: The repository 'http://deb.debian.org/debian-security buster/updates Release' does not have a Release file.
E: The repository 'http://deb.debian.org/debian buster-updates Release' does not have a Release file.
```

It can still be found on https://archive.debian.org/debian-archive/ and that would need to be added to the apt sources for it to work.

Who knows how to hack that into the `.github/workflows/ci.yml` file?

rene-dev, 2025-07-20:
fixed by 7256a6599059c4303cc04f05fdab0319c2d9a37e

closed 2025-07-20 by rene-dev

BsAtHome, 2025-07-20:
I don't think it is "solved" by simply removing the distro version from CI. The ELTS status is that there is (paid) support up til 2029.

The real problem is that Buster is a supported base for LinuxCNC 2.9 branch. There are several package version things that need to be tested with Buster when things are committed (f.ex. old python).

rene-dev, 2025-07-20:
Its end of life, and linuxcnc runs fine on newer versions of Debian. We do not have the manpower to support legacy os versions. If you have the time and energy to find a better solution, feel free to create a PR.