LinuxCNC/linuxcnc #3538: PyVCP Widgets param_pin + halparam - Error: "object has no attribute 'param_pin'"

andronick83, 2025-08-06:
```
$ cat test.xml
<pyvcp>
	<spinbox halpin="sb-0" param_pin="1" halparam="sb-0-in"/>
	<spinbox halpin="sb-1" param_pin="1"/>
	<spinbox halpin="sb-2"/>
	<spinbox/>
</pyvcp>
```

```
$ pyvcp test.xml
Error constructing spinbox({'halpin': 'sb-1', 'param_pin': 1, 'halparam': 'sb-1-in'}):
'pyvcp_spinbox' object has no attribute 'param_pin'
```

[pyvcp_widgets.py:982](https://github.com/LinuxCNC/linuxcnc/blob/master/lib/python/pyvcp_widgets.py#L982)

```
        if halparam == None:
            self.param_pin = param_pin
            if self.param_pin == 1:
                halparam = "spinbox." + str(pyvcp_spinbox.n) + ".param_pin"
                self.halparam=halparam
                pycomp.newpin(halparam, HAL_FLOAT, HAL_IN)
```
Replace with:
```
        self.param_pin = param_pin
        if self.param_pin == 1:
            if halparam == None:
                halparam = "spinbox." + str(pyvcp_spinbox.n) + ".param_pin"
            self.halparam=halparam
            pycomp.newpin(halparam, HAL_FLOAT, HAL_IN)
```

Result:
```
$ pyvcp test.xml &
[1] 1584156
$ halcmd show pin 
Component Pins:
Owner   Type  Dir         Value  Name
     2  float OUT             0  test.sb-0
     2  float IN              0  test.sb-0-in
     2  float OUT             0  test.sb-1
     2  float OUT             0  test.sb-2
     2  float IN              0  test.spinbox.1.param_pin
     2  float OUT             0  test.spinbox.3
```

andypugh, 2025-08-06:
Which LinuxCNC version? Do you think that this has always been a problem? (It just seems odd that it's taken 20 years to find it.)

Your XML is formatted very differently from the documented examples, does that make any difference? 

https://linuxcnc.org/docs/stable/html/gui/pyvcp.html

andypugh, 2025-08-06:
I admit to being puzzled, as this area of the code has been stable for 12 years. I don't suppose it can be a python2 / python3 thing? It doesn't have that look to it, the code looks compatible with both.

andypugh, 2025-08-06:
I did a search for files in the sample configs that use the param_pin.  

[configs/sim/pyvcp_demo/custom.hal]
[configs/sim/axis/gantry/gantry_panel.hal]
[configs/sim/axis/anglejog/anglejog_postgui.hal]
[configs/sim/axis/anglejog/anglejog.xml][configs/sim/axis/gantry/gantry_panel.xml]
[configs/sim/pyvcp_demo/pyvcp_embed_tab.hal]
[configs/sim/pyvcp_demo/pyvcp_embed_tab_1.xml]
[configs/sim/pyvcp_demo/pyvcp_bottom_panel.hal]
[configs/sim/pyvcp_demo/pyvcp_bottom_panel.xml]
[configs/sim/axis/external_offsets/jwp_z_panel.xml][configs/sim/axis/external_offsets/dynamic_offsets_panel.xml]
[configs/apps/pyvcp/pyvcp_demo1.xml]
[configs/sim/axis/external_offsets/eoffsets_panel.xml]
[configs/apps/pyvcp/pyvcp_demo1.hal]
[configs/sim/axis/moveoff/moveoff_display_7.xml]
[configs/sim/axis/external_offsets/opa_panel.xml]
[configs/sim/pyvcp_demo/pyvcp_embed_tab_2.xml]
[configs/sim/pyvcp_demo/pyvcp_demo1.xml]
[configs/apps/pyvcp/pyvcp_demo1_es.xml]

I have only tried a few so far, but they all seem to work. 

I am not trying to argue that there isn't a problem, but we need to define the boundaries of the problem to fix it. 

So, first of all, does configs/apps/pyvcp/pyvcp_demo1 work for you? It does work for me on a stock LinuxCNC 2.9.4 ISO install in a VM.

andronick83, 2025-08-07:
The linuxcnc_2.9.4-amd64.hybrid.iso image is installed
Yes, all the XMLs you listed work for me too.
Also works "configs/apps/pyvcp/pyvcp_demo1"

All the XMLs listed do not use "halparam". I did not find any XML on the system that uses it.
configs/common/configurable_options/pyvcp/m5i20panel.xml also does not seem to use "halparam", but it causes a similar error. Maybe for another reason:

m5i20panel.xml:1591
```xml
    <dial>
        <size>125</size>
        <cpr>100</cpr>
        <text>"Jog"</text>
        <initval>0</initval>
        <resolution>1</resolution>
        <halpin>"anaout.0"</halpin>
        <dotcolor>"black"</dotcolor>
    </dial>
```
Pyvcp doesn't like something about this "dial"

I don't remember where I found it. Maybe I just looked into pyvcp_widgets.py to find out all the widgets' capabilities.
Without "halparam" param_pin creates numbered pins like:
pyvcp.dial.0.param_pin
pyvcp.dial.1.param_pin
Which is inconvenient for tracking.
With "halparam" working, pins can be named in their own way, which seemed more convenient to me.

andronick83, 2025-08-08:
Almost. You forgot to delete the lines I commented out:
[L201-L206](https://github.com/LinuxCNC/linuxcnc/blob/795a04abc895c9f41a6f2be87b9abb3bd5622e67/lib/python/pyvcp_widgets.py#L201-L206) [L987-L992](https://github.com/LinuxCNC/linuxcnc/blob/795a04abc895c9f41a6f2be87b9abb3bd5622e67/lib/python/pyvcp_widgets.py#L987-L992) [L1574-L1579](https://github.com/LinuxCNC/linuxcnc/blob/795a04abc895c9f41a6f2be87b9abb3bd5622e67/lib/python/pyvcp_widgets.py#L1574-L1579)

Correct commit [8a4d8cca](https://github.com/andronick83/linuxcnc/pull/1/commits/8a4d8ccaa57bff7a9a0161ee86711f59026098bc)

andypugh, 2025-08-08:
Can you create this as a pull-request into LinuxCNC 2.9? 

https://github.com/LinuxCNC/linuxcnc/pulls

commit 8c7016f0 (2025-08-24): Fix #3538 - "halparam" tag not working in PyVCP

https://github.com/LinuxCNC/linuxcnc/issues/3538
commit d49b49ab (2025-08-24): Merge pull request #3540 from andronick83/patch-1

Fix #3538 - "halparam" tag not working in PyVCP
closed 2025-08-15 by andypugh