LinuxCNC/linuxcnc #3563: 2.9.5 axis - touch off with "end" key pressed: not possible to input any numbers then 0 - O.K. is greyed out

dd3t, 2025-09-15:
2.9.5 axis - touch off with "end" key pressed: not possible to input any numbers then 0 - O.K. is greyed out

[image]

https://forum.linuxcnc.org/21-axis/57208-touch-off-ok-button-greyed-out

andypugh, 2025-09-16:
I have done some digging, but remain confused. 

The touch-off entry box has a callback to check_valid that is called whenever the entry changes:
https://github.com/LinuxCNC/linuxcnc/blob/2.9/src/emc/usr_intf/axis/scripts/axis.py#L1688
This, in turn calls parse_gcode_expression:
https://github.com/LinuxCNC/linuxcnc/blob/2.9/src/emc/usr_intf/axis/scripts/axis.py#L1580
Which attempts to load a parameter file. 
But I am not sure why, as check_valid checks for the # character and rejects it saying that variables can not be used. 
I suspect that this "File not open" error is set here:
https://github.com/LinuxCNC/linuxcnc/blob/2.9/src/emc/rs274ngc/rs274ngc_pre.cc#L1602
As described:
https://github.com/LinuxCNC/linuxcnc/blob/2.9/src/emc/rs274ngc/rs274ngc_pre.cc#L1485

I think that we may just have been getting away with this, as the behaviour removed in #2753 was setting one of the pointers referenced non-null. 

But in the case of checking a G-code expression, that's perfectly fine, and we can probably just add in special handling for error 4, 
INTERP_FILE_NOT_OPEN that treats it as not-an-error

andypugh, 2025-09-16:
Changing line 1590 in axis.py to
```py
    # The "4" should be INTERP_FILE_NOT_OPEN 
    if result > 4: return False, gcode.strerror(result)
```
Seems to bypass the problem. 
But I am not sure if this is a proper fix for a not-really-a-problem or a kludge.

Sigma1912, 2025-09-17:
After some more digging into the original issue https://github.com/LinuxCNC/linuxcnc/issues/2753 it looks like I missed that the RS274NGC_STARTUP_CODE is inserted in 'gremlin.py':
https://github.com/LinuxCNC/linuxcnc/blob/master/src/emc/usr_intf/gremlin/gremlin.py#L330
Which would indicate that I was trying to fix this in the wrong place.

I would thus suggest to revert https://github.com/LinuxCNC/linuxcnc/commit/736a3a904aaa70532a7ee83ef1c196853ab7ba42 and I'll see if I can find a better fix for https://github.com/LinuxCNC/linuxcnc/issues/2753

Sigma1912, 2025-09-17:
My appologies, in hindsight, [this](https://github.com/LinuxCNC/linuxcnc/commit/736a3a904aaa70532a7ee83ef1c196853ab7ba42) was a rather half baked fix.

Sigma1912, 2025-09-18:
@andypugh 
Do you want to revert the merge of https://github.com/LinuxCNC/linuxcnc/commit/736a3a904aaa70532a7ee83ef1c196853ab7ba42 or should I make a commit in my new PR addressing https://github.com/LinuxCNC/linuxcnc/issues/2753?

closed 2025-09-19 by andypugh