LinuxCNC/linuxcnc #3564: Synchronize EtherCAT and linuxcnc clocks

MetalMusings, 2025-09-15:
I would like to see LinuxCNC's clock run synchronously with EtherCAT's DC (distributed clock).
As it is now, one of the clocks always runs a little bit faster which, together with latency, causes noise
and intermittent high accelerations and torque of EtherCAT drives using DC which is all servo and stepper
motor drives. Some drives are better than others to absorb this effect, but it is always there.

This is caused by LinuxCNC sending data to the EtherCAT master via the lcec module either too fast or too slow (not synchronized).
What reaches the drives, when the problem occurs, is either no new position data or position data not for the next LinuxCNC servo cycle but for the next-next cycle.
This causes the drive to either brake hard or accelerate hard for one cycle. Hence the noise. Sounds like gravel in the drive.

The realistic way to improve on this is to have LinuxCNC's clock adjust or synchronize itself with EtherCAT's DC.
Or use EtherCAT's clock as the time source for LinuxCNC.
It's not realistic to have DC adjust itself to LinuxCNC's clock. It's a closed system. The first DC-capable slave on the EtherCAT bus
sets the clock speed and all other slaves adjust to that clock. The EtherCAT master also has the DC time, 
and can be made accessible via a modified lcec module to LinuxCNC, or any other way as needed.

satiowadahc, 2025-09-15:
How I did it with an FC1121 is used a hal pin as an interrupt to poll data from Ethercat. 

Another way I have seen it improved when LCEC is the master, (Not a slave) is to improve your system latency. This guy has a good write up: dantalion.nl/2024/09/29/linuxcnc-latency-jitter-kernel-parameter-tuning.html

(Just noting officially LCEC and linuxcnc are separate projects.) 
https://github.com/linuxcnc-ethercat/linuxcnc-ethercat/issues

andypugh, 2025-09-17:
Anything useful here? 
https://github.com/machinekit/machinekit/issues/687

RTAI might support external clock. PREEMPT-RT might too, but I have not even hear rumours.

MetalMusings, 2025-09-17:
Absolutely, thanks Andy. Very relevant. A lot to digest though.

MetalMusings, 2025-09-19:
There is provision made for adjusting the real-time clock, although I don't see it used anywhere.
It's the task->pll_correction

Creation of servo-thread
https://github.com/LinuxCNC/linuxcnc/blob/52b35be1b542184e7a7b30bf63895a33e6d15e5e/src/emc/motion/motion.c#L1014
https://github.com/LinuxCNC/linuxcnc/blob/52b35be1b542184e7a7b30bf63895a33e6d15e5e/src/hal/hal_lib.c#L2154
https://github.com/LinuxCNC/linuxcnc/blob/52b35be1b542184e7a7b30bf63895a33e6d15e5e/src/rtapi/uspace_rtapi_app.cc#L1239

Starting the task
https://github.com/LinuxCNC/linuxcnc/blob/52b35be1b542184e7a7b30bf63895a33e6d15e5e/src/rtapi/uspace_rtapi_app.cc#L1248
https://github.com/LinuxCNC/linuxcnc/blob/52b35be1b542184e7a7b30bf63895a33e6d15e5e/src/rtapi/uspace_rtapi_app.cc#L1083
https://github.com/LinuxCNC/linuxcnc/blob/52b35be1b542184e7a7b30bf63895a33e6d15e5e/src/rtapi/uspace_rtapi_app.cc#L1118
And there is the task->pll_correction that can be manipulated

During operation it looks like rtapi_wait() is used to wait using nanosleep until it is time to wake up and execute the task
https://github.com/LinuxCNC/linuxcnc/blob/52b35be1b542184e7a7b30bf63895a33e6d15e5e/src/rtapi/uspace_rtapi_app.cc#L1285
https://github.com/LinuxCNC/linuxcnc/blob/52b35be1b542184e7a7b30bf63895a33e6d15e5e/src/rtapi/uspace_rtapi_app.cc#L1161
https://github.com/LinuxCNC/linuxcnc/blob/52b35be1b542184e7a7b30bf63895a33e6d15e5e/src/rtapi/uspace_rtapi_app.cc#L1171
task->pll_correction is there again.

There is also a function made to set pll_correction
https://github.com/LinuxCNC/linuxcnc/blob/52b35be1b542184e7a7b30bf63895a33e6d15e5e/src/rtapi/uspace_rtapi_app.cc#L1133

So the strategy must be to make pll_correction available for adjustment, and adjust it.

MetalMusings, 2025-09-19:
Looking in lcec_main.c there IS synchronization of clocks.
https://github.com/linuxcnc-ethercat/linuxcnc-ethercat/blob/bf13577ec7b1b54844e07a80d949232e71776b32/src/lcec_main.c#L1213

There is a limit of +-1% that can be corrected, though.
https://github.com/LinuxCNC/linuxcnc/blob/52b35be1b542184e7a7b30bf63895a33e6d15e5e/src/rtapi/uspace_rtapi_app.cc#L1051

A quick test is to increase this 1% to, say, 10% and possibly se a difference.

MetalMusings, 2025-09-20:
To activate  synchronization, this line must evaluate to true
https://github.com/linuxcnc-ethercat/linuxcnc-ethercat/blob/bf13577ec7b1b54844e07a80d949232e71776b32/src/lcec_main.c#L1176
That is, the RefClockSyncCycles value must be negative, as in 
`  <master idx="0" appTimePeriod="1000000" refClockSyncCycles="-10">`
I have tested it and made some measurements, and indeed the linuxcnc clock drifts slightly. Over 5 minutes the pll mechanism corrected some 40 microseconds. How much depends on the accuracy of the first slave's clock.

I will close this issue.
Linuxcnc is prepared for synchronization with ethercat DC and will do it when enabled.
LCEC enables synchronization when RefClockCyncCycles="negative value".
This is not described anywhere in the documentation, I found out from the code.
I will open an issue with LCEC to have this added to the documentation.